praveen-palanisamy/multiple-object-tracking-lidar #2: How to this problem

PujieXin, 2018-07-27:
I think I configure the environment right, however, I meet the errors:
/catkin_ws/src/multiple-object-tracking-lidar/src/main.cpp:355:28: error: no match for ‘operator*’ (operand type is ‘cv::MatCommaInitializer_<float>’)
     KF0.transitionMatrix = *(Mat_<float>(4, 4) << dx,0,1,0,   0,dy,0,1,  0,0,dvx,0,  0,0,0,dvy);
error: no match for ‘operator*’ (operand type is ‘cv::MatCommaInitializer_<float>’)
     KF1.transitionMatrix = *(Mat_<float>(4, 4) << dx,0,1,0,   0,dy,0,1,  0,0,dvx,0,  0,0,0,dvy);

How can I solve it? Thank you!

praveen-palanisamy, 2018-07-30:
Which version of OpenCV are you using?

With your OpenCV version it looks like, the `Mat_` is probably not returning a pointer as it used to be. 
You should try removing the indirection operator (`*`)  by changing the following Lines:
https://github.com/praveen-palanisamy/multiple-object-tracking-lidar/blob/8bd23a2ee7325f449bbd746589348c90ffb4de7d/src/main.cpp#L355-L360

to 

```c++
KF0.transitionMatrix = (Mat_<float>(4, 4) << dx,0,1,0,   0,dy,0,1,  0,0,dvx,0,  0,0,0,dvy);
KF1.transitionMatrix = (Mat_<float>(4, 4) << dx,0,1,0,   0,dy,0,1,  0,0,dvx,0,  0,0,0,dvy);
KF2.transitionMatrix = (Mat_<float>(4, 4) << dx,0,1,0,   0,dy,0,1,  0,0,dvx,0,  0,0,0,dvy);
KF3.transitionMatrix = (Mat_<float>(4, 4) << dx,0,1,0,   0,dy,0,1,  0,0,dvx,0,  0,0,0,dvy);
KF4.transitionMatrix = (Mat_<float>(4, 4) << dx,0,1,0,   0,dy,0,1,  0,0,dvx,0,  0,0,0,dvy);
KF5.transitionMatrix = (Mat_<float>(4, 4) << dx,0,1,0,   0,dy,0,1,  0,0,dvx,0,  0,0,0,dvy);
```

Try it out and let me know if that fixes your issue.

PujieXin, 2018-08-01:
It is solved, thank you very much!

closed 2018-08-01 by PujieXin