praveen-palanisamy/multiple-object-tracking-lidar #4: Deleting unused markers

THAC00, 2018-08-11:
Hey Praveen! 
Amazing work, it has been very useful helping me to figure out a way to associate cluster data between frames.
Currently I'm trying to find out a way to remove markers (at least out of sight) , which are currently not tracking any extracted cluster.

I played around a bit with the code, but I can't get it to work, maybe you have some ideas ? 
Also did you figure out a way, to draw "tracking lines", following the cluster, instead of simple boxes in the cluster center?

Thank you in Advance!
best regards

praveen-palanisamy, 2018-08-13:
Hey @THAC00 ,

Very glad to know that this code repo has been useful for you.

The Markers are published for up-to 6 cluster locations resulting from the clustering process. So, each marker is placed at each of the cluster centers using the following lines in the code: https://github.com/praveen-palanisamy/multiple-object-tracking-lidar/blob/8bd23a2ee7325f449bbd746589348c90ffb4de7d/src/main.cpp#L242-L269

I am not sure based on what criteria you want to filter out the markers but you can modify the above lines to publish only those clusters that are of interest to you. For example, you can choose to have a (distance) threshold to associate the clusters to a KF tracker and only display/publish markers for those clusters that satisfy the threshold.

THAC00, 2018-08-13:
Thanks for the reply !  The idea is to only show markers which are actively tracking a cluster . Right now , when you lose a tracker , the marker will stay in place of the last position.  So I'm wondering how to identify if a tracker is " active " and make an if clause to remove the marker if it isn't

praveen-palanisamy, 2018-08-13:
Regarding your second question about "tracking lines",  you can use the `visualization_msgs::Marker::POINTS`  and `visualization_msgs::Marker::LINE_STRIP` marker types instead of the `visualization_msgs::Marker::CUBE` used in this this line:https://github.com/praveen-palanisamy/multiple-object-tracking-lidar/blob/8bd23a2ee7325f449bbd746589348c90ffb4de7d/src/main.cpp#L249
for the markers to create that "tracking lines" that show the trajectories of the tracked clusters.

You will have to make some modifications to the code. It will be something like this:

```c++

// Assuming the `prevClusterCenters[i][j]` is a 2D matrix that has the
//  `j^th` previous cluster center for the `i^ith` KFprediction,
for (int i = 0; i < 6; i++)
{
    visualization_msgs::Marker points, line_strip;
    points.header.frame_id = line_strip.header.frame_id = "/map";

    // This is where we are changing the marker type from BOX to POINTS & LINE_STRIP
    points.type = visualization_msgs::Marker::POINTS;
    line_strip.type = visualization_msgs::Marker::LINE_STRIP;

    // Set the marker action.  Options are ADD and DELETE   
    points.action = line_strip.action = visualization_msgs::Marker::ADD;
    
    for(int j=0; j < numPreviousPredictions; j++)
    {
       geometry_msgs::Point p;
       p.x = prevClusterCenters[i][j].x;
       p.y = prevClusterCenters[i][j].y;
       points.push_back(p);
       line_strip.points.push_back(p);
    }
    markerPub.publish(points);
    markerPub.publish(line_strip);
}
```
Hope that helps.

THAC00, 2018-08-13:
Thanks a lot for the input ! I appreciate it. The code happens to be the only openly available documentation of point cloud cluster tracking with pcl !

praveen-palanisamy, 2018-08-13:
> Thanks for the reply ! The idea is to only show markers which are actively tracking a cluster . Right now , when you lose a tracker , the marker will stay in place of the last position. So I'm [...]

Oh I see. That's mainly because the code currently looks for a fixed (6) number of clusters. That's why when there are less than 6 actively tracked clusters, you see stationary markers on the old clusters until new objects are being tracked. To overcome that, as mentioned in the code comment, which you quoted in the edited version of the issue description, you can dynamically decide how many KF trackers to initialize based on the number of detected object clusters in the point cloud. 

It's been quite a while since I worked on this and I do not have a version with me that is generic and dynamically creates a tracker instance based on the number of detected object point clouds. I currently do not have the bandwidth to work on it but I will be happy to assist or merge a pull request (PR).

THAC00, 2018-08-13:
Thanks a ton! I'll work on it an keep you updated . Maybe lastly, if time allows, I've opened another issue asking on how to retrieve the velocities and convert them to m/s, because right now I can't make sense of the unit. Thanks in advance and best regards

closed 2018-08-13 by THAC00