praveen-palanisamy/multiple-object-tracking-lidar #6: how to run your project

varshadevi2018, 2019-03-13:
can you tell the way to run your project... after caktin_make what should I do

praveen-palanisamy, 2019-04-13:
@varshadevi2018 :
Listed below are the necessary steps to use this (`multi_object_tracking_lidar`) ROS package:

1. [Create a catkin workspace](http://wiki.ros.org/catkin/Tutorials/create_a_workspace) (if you do not have one setup already). 
1. Navigate to the `src` folder in your catkin workspace: `cd ~/catkin_ws/src`
1. Clone this repository: `git clone https://github.com/praveen-palanisamy/multiple-object-tracking-lidar.git`
1. Compile and build the package: `cd ~/catkin_ws && catkin_make`
1. Add the catkin workspace to your ROS environment: `source ~/catkin_ws/devel/setup.bash`
1. Run the `kf_tracker` ROS node in this package: `rosrun multi_object_tracking_lidar kf_tracker`

If all went well, the ROS node should be up and running! As long as you have the point clouds (from LIDAR or other pointcloud generator) published on to the `filtered_cloud` rostopic, you should see outputs from this node published onto the `obj_id`, `cluster_0`, `cluster_1`, …, `cluster_5` topics along with the markers on `viz` topic which you can visualize using RViz.

I have added these steps to the [README here](https://github.com/praveen-palanisamy/multiple-object-tracking-lidar#usage) as well. Let me know if you are/aren't able to run it.

praveen-palanisamy, 2019-04-19:
Hey @varshadevi2018 : I'm closing this as your question has been addressed. Feel free to re-open or open a new issue if something is unresolved.

closed 2019-04-19 by praveen-palanisamy