praveen-palanisamy/multiple-object-tracking-lidar #8: Moving platform

YangSiri, 2019-04-18:
Hi, i think you've done a great job with this simple lidar tracker. Here i'd like to raise a question about its adaptation on outdoor moving platform. Is it robust ?

praveen-palanisamy, 2019-04-19:
Hey @YangSiri : Thank you. The algorithms used are applicable to outdoor moving platform too. In fact, this was developed mainly for outdoor autonomous navigation application. Though, I haven't tested this extensively at very high speeds.

YangSiri, 2019-04-20:
That's great. Thanks for your reply. I'm gonna try it.

wanfuse123, 2019-04-28:
So I installed this and ROS and everything and started ROS. Got to this point

Run the kf_tracker ROS node in this package: rosrun multi_object_tracking_lidar kf_tracker

works ok? So  now what do I do? 

Right now I have no real data but the two scenarios I am looking at are

1) what do i do next when I do get a lidar?

2) how do I continue to run it on test data and actually see the output?

Thanks in advance for your help!

praveen-palanisamy, 2019-05-14:
hey @wanfuse123 : Glad that you are able to run the ROS node. 
1. Run the node on your robot/platform on which you will be mounting the LIDAR to detect/cluster and track obstacles/objects.
2. You can also just use logged LIDAR data (eg. Played from a rosbag file) to test. Once you have the `kf_tracker` node running, as long as you have the point clouds (from LIDAR or other pointcloud generator) published on to the `filtered_cloud` rostopic, you should see outputs published onto the `obj_id`, `cluster_0`, `cluster_1`, …, `cluster_5` topics along with the markers on `viz` topic which you can visualize using RViz.