praveen-palanisamy/multiple-object-tracking-lidar #14: ROS node hangs

VijeetBenni, 2019-08-27:
I followed the steps from the ReadMe, when I run the tracker by below command, (with roscore running)
**_$rosrun multi_object_tracking_lidar kf_tracker_**
It hangs showing **About to setup callback**

Do I need to do something else? Like providing inputs ? From where does it takes the input?

VijeetBenni, 2019-08-27:
Thanks Praveen.

Since I am new to ROS, not aware of ros commands.
Can you direct me on how to use rostopic etc.,?

xph123123, 2019-09-04:
Thanks for your wonderful contribution, i came through the same problem as "It hangs showing About to setup callback".
Then, i changed the subscribed topic in main.cpp from "filtered_cloud" to "pandar_points" which is the topic published by the lidar i used. but it still didn't work.
i can't figure it out.

xph123123, 2019-09-04:
The "pandar_points" is in the form of PointCloud2

praveen-palanisamy, 2019-09-05:
@VijeetBenni : There are a lot of tutorials and documentation online that cover how to publish data to ROS topics. Please refer to them. Some pointers:
1. [Understanding ROS topics](http://wiki.ros.org/ROS/Tutorials/UnderstandingTopics)
2. [Laser scans and point clouds in ROS; Converting laser scans to point clouds](http://wiki.ros.org/laser_pipeline/Tutorials/IntroductionToWorkingWithLaserScannerData)

@xph123123 : You can use the [remap tag](https://wiki.ros.org/roslaunch/XML/remap) to remap "pandar_points" to "filtered_cloud".

If you change the topic anme in the code, please make sure you rebuild this ROS node before running to make sure your code change came into effect.

xph123123, 2019-09-07:
Thank you very much for your help. The problem has been solved!

praveen-palanisamy, 2019-09-07:
@xph123123 : Glad that your problem has been solved!

Closing this issue. Reopen or create a new one if required.

closed 2019-09-07 by praveen-palanisamy

Tonyc48, 2021-04-06:
@praveen-palanisamy: I have the same issue.  Do I need to publish each cluster, obj_id, and viz to the filtered_cloud topic?  After running "rosrun multi_object_tracking_lidar kf_tracker", I look at the rostopic list and it reads:
Published topics:
 * /rosout_agg [rosgraph_msgs/Log] 1 publisher
 * /rosout [rosgraph_msgs/Log] 2 publishers
 * /cluster_0 [sensor_msgs/PointCloud2] 1 publisher
 * /cluster_1 [sensor_msgs/PointCloud2] 1 publisher
 * /cluster_2 [sensor_msgs/PointCloud2] 1 publisher
 * /cluster_3 [sensor_msgs/PointCloud2] 1 publisher
 * /cluster_4 [sensor_msgs/PointCloud2] 1 publisher
 * /cluster_5 [sensor_msgs/PointCloud2] 1 publisher
 * /obj_id [std_msgs/Int32MultiArray] 1 publisher
 * /viz [visualization_msgs/MarkerArray] 1 publisher
 * /initialpose [geometry_msgs/PoseWithCovarianceStamped] 1 publisher
 * /move_base_simple/goal [geometry_msgs/PoseStamped] 1 publisher
 * /clicked_point [geometry_msgs/PointStamped] 1 publisher

Subscribed topics:
 * /rosout [rosgraph_msgs/Log] 1 subscriber
 * /filtered_cloud [sensor_msgs/PointCloud2] 2 subscribers
 * /tf [tf2_msgs/TFMessage] 1 subscriber
 * /tf_static [tf2_msgs/TFMessage] 1 subscriber

I even went into the main file of kf_tracker source code and included:
 #include <pcl_ros/point_cloud.h>
I still get the "About to setup callback" error.

I also opened rviz and tried to add the PointCloud2 display with filtered_cloud topic.  I've tried just about everything but nothing seems to work.  I would really love your help!