praveen-palanisamy/multiple-object-tracking-lidar #15: Hi I published data on filtered_cloud but now getting this error

sab3rclaw, 2019-10-02:
About to setup callback
kf_tracker: /build/pcl-6_P28C/pcl-1.7.2/kdtree/include/pcl/kdtree/impl/kdtree_flann.hpp:172: int pcl::KdTreeFLANN<PointT, Dist>::radiusSearch(const PointT&, double, std::vector<int>&, std::vector<float>&, unsigned int) const [with PointT = pcl::PointXYZ; Dist = flann::L2_Simple<float>]: Assertion `point_representation_->isValid (point) && "Invalid (NaN, Inf) point coordinates given to radiusSearch!"' failed.
Aborted (core dumped)

firesnow1234, 2019-10-22:
My question is the same as yours.
Did you have solved it?

praveen-palanisamy, 2019-10-26:
Hi @sab3rclaw : As the error message indicates, you seem to be publishing invalid (NaN/Inf) point data on `filtered_cloud`.
Verify that the point cloud data you publish has only valid point cloud data. You can print the points to console (`std::cout`) to verify the issue.
To fix the issue in your publisher code, you can `#include <pcl/filters/filter.h>` and then use `pcl::removeNaNFromPointCloud (...)`

closed 2019-10-26 by praveen-palanisamy