praveen-palanisamy/multiple-object-tracking-lidar #16: Background of the dataset used

jadhavhninad, 2019-10-24:
Hi Praveen,
Thanks for the great project! I have the following questions : 
1. Can you give some more context as to which data did you use? (the one in the image). 
2. Does it limit to tracking only 5 objects or can that parameter be changed? 
3. When visualizing in rviz, how do I overlay the markers with the pointcloud? It seems that the filtered_cloud uses velodyne frame  while the markers use map frame.

praveen-palanisamy, 2019-10-26:
Hey @jadhavhninad  please find responses to your questions below:

1. The data used in the image demo was collected from a wheeled, mobile robot with a LiDAR
2. It can be changed to track more than 5 objects. I only needed to track 5 objects at most. It can be extended to `N` object tracking.
3. The point clouds are generated w.r.t the Velodyne sensor's frame so, it is appropriate to visualize the points w.r.t to that frame of reference. Depending on your robot/setup you can change the  LiDAR sensor's coordinate frame (to map). You could also convert the points from the map's frame to the Velodyne's frame.

praveen-palanisamy, 2019-11-21:
@jadhavhninad : Hope the above responses were adequate. Closing this item. Feel free to re-open or create a new one if you need more info.

closed 2019-11-21 by praveen-palanisamy