praveen-palanisamy/multiple-object-tracking-lidar #17: obstacle detection problem

Zhang-Qinghui, 2019-11-14:
hi,thanks for your work.Does the project  only detect dynamic obstacles?

praveen-palanisamy, 2019-11-14:
Hi @Zhang-Qinghui , thank you for your interest in this project. The code supports detection, classification and tracking of both dynamic and static obstacles.

Zhang-Qinghui, 2019-11-14:
Thanks for your reply. I got it. If I fix the velodyne, How can I modify the code to only detect dynamic obstacles.

praveen-palanisamy, 2019-11-14:
You can apply a filter over the detected objects to remove objects whose velocity is zero. In other words, if there is no change in the position of an objects in `n` consecutive frames, you can treat that to be stationary and filter it out.

Zhang-Qinghui, 2019-11-14:
what you said is meaningful.Thank you.

Zhang-Qinghui, 2019-11-14:
But I can not find the direction arrow as your picture shows when I see the reault through rviz. I can see cluster_0, cluster_1, …, cluster_5 topics and the markers on viz topic through rviz. How can I see the direction arrow as your picture shows? thank you in advance.
[image]

praveen-palanisamy, 2019-11-16:
The markers on the viz topic should look similar on RViz as shown in this project's repo README depending on your point cloud data.

Can you share a screenshot of your visualization to see what's missing in your RViz?

Zhang-Qinghui, 2019-11-17:
Thanks for your reply.There is the screenshot of my visualization.In additon,only the Fixed Frame set as 'map' can I see the marker.
[image]

praveen-palanisamy, 2019-11-17:
I see. So, the objects from the point clouds are being detected, clustered and tracked. The missing piece is that the point clouds are not shown on RViz which is why you don't see the additional information. Please make sure that the point cloud data is published on to the **filtered_cloud** ROS topic and in RViz, choose that topic for "PointCloud2". You should then see all the information. 

Yes. The **map** "FIced Frame" sounds right.

Zhang-Qinghui, 2019-11-17:
Maybe I do not say clear. I mean, I can't see the arrow as your picture shows, just as the picture shows below .
[image]

praveen-palanisamy, 2019-11-17:
The "arrow" like line segments are derived from the point clouds and will only show up when the point cloud data is streamed into RViz (using the steps in my previous comment).

Zhang-Qinghui, 2019-11-17:
ok,thank you.

closed 2019-11-21 by praveen-palanisamy