praveen-palanisamy/multiple-object-tracking-lidar #18: Detecting/Segmenting static objects

abhigoku10, 2020-01-22:
@praveen-palanisamy  hi thanks for opensourcing your code i have few queries 
1. i need to detect/segment static objects like trees / light post / traffic lights/buildings is it possible 
2. if so what method are  you using since using the static objects i need to localize the position 
thanks in advance

praveen-palanisamy, 2020-01-27:
Yes. You can use the `obj_ids` and the `cluster_i` output by this package to detect/segment the static objects of interest.
The `cluster_i` will give the position of a segmented object to help you localize the object in the given scene.

praveen-palanisamy, 2020-03-22:
Closing this assuming the response above addressed your question. Re-open or create a new issue if you need more help.

closed 2020-03-22 by praveen-palanisamy