praveen-palanisamy/multiple-object-tracking-lidar #19: Retrieval of covariance matrices for data association

Shivarama123, 2020-01-24:
Can we get the covariance matrices for data association from your algorithm ?? .... If yes can you please let me know how to retrieve the covariance matrices.

Thanks in advance!

praveen-palanisamy, 2020-01-27:
Hi @Shivarama123 ,

The covariance matrices are not currently published but, you could easily publish them to a new topic as they can be accessed from the corresponding Kalman Filter tracker object (e.g. KF0) using the following reference table:

|                      Data                     | Attribute to get the value |
|:---------------------------------------------:|----------------------------|
| Measurement Noise Covariance                  | KF0.measurementNoiseCov    |
| Process Noise Covariance                      | KF0.processNoiseCov        |
| Posterior Error Estimate Covariance | KF0.errorCovPost           |
|                                               |                            |

Hope that helps.

Shivarama123, 2020-01-29:
Thank you so much for your reply. 

And regarding data association, can you please let me know which of the covariance matrices should be required?

Only the diagonal elements of Process noise covariance will suffice or do I have to consider some other as well?

It will be a great help. Thank you so much for your time.

praveen-palanisamy, 2020-02-01:
It depends on the data association algorithm/process that you are trying to implement/use. 
You can start with the process or measurement noise covariance.

praveen-palanisamy, 2020-03-22:
Closing this as the questions were answered. Re-open if you have any related follow-up Qs or open a new item.

closed 2020-03-22 by praveen-palanisamy