praveen-palanisamy/multiple-object-tracking-lidar #20: UAV tracking

cacao1987, 2020-03-18:
We plan to install lidar on the tower to track unmanned aerial vehicles in the air. Can we use this algorithm? Looking forward to your reply.

praveen-palanisamy, 2020-03-22:
Hi @cacao1987 , thank you for your interest. Good to hear about your use case.
If you have a sample LIDAR scan from your setup, it will be very helpful to say whether the code in this repository will work well for your use case. 

If you don't have access to a sample LIDAR scan yet, then, based on your description, I am guessing that you want to detect and track multiple aerial vehicles which typically may be (?) small in size (depending on the height of your tower and the flying heights of the other UAVs) and therefore could have very small number of LIDAR points per object/UAV. In such a case, as long as there are not multiple UAVs very close to each other, you might still get detection and tracking results from this implementation as is but, you may get false positives if the number of points are too small and sparse per UAV (i.e they are too far).

The code is easy enough to be extended to suit your particular use case for the best performance.
Hope that helps.

cacao1987, 2020-03-23:
@praveen-palanisamy  thanks for youe reply, in the next stage, we will use lidar scanning data to verify

praveen-palanisamy, 2020-04-09:
Closing this now. Please reopen when you are in the next stage of your project and need help.

closed 2020-04-09 by praveen-palanisamy