praveen-palanisamy/multiple-object-tracking-lidar #21: Unsupervised k-means

tberrielITA, 2020-03-23:
Hello, I am interested on the unsupervised k-means clustering, but I can't get to find where you are using it. I've explored the code and observed that the main.cpp uses only the Kd-tree search and Kalman filter, but there isn't a single call to a kmeans function. Also it looks like featureDetection.cpp is for image processing and not for point clouds. 

Where can i find the unsupervised kmeans implementation you talk about on the ReadMe?

Thank you for your help.

praveen-palanisamy, 2020-03-27:
Hi @tberrielITA ,

K-means is not used in the case of 3D point clouds. The default code flow is setup for 3D point clouds and therefore k-means isn't used while the code is available in the repo (`featureDetection.cpp`) if someone needs to obtain similar results on a 2D input.

In the 3D point-cloud (default) case, instead of k-means, [Euclidean cluster extraction](http://mediatum.ub.tum.de/doc/800632/941254.pdf) is performed in these lines using PCL:
https://github.com/praveen-palanisamy/multiple-object-tracking-lidar/blob/d71ff08c2eda1be00395468a4243c011d1d98729/src/main.cpp#L409-L410  

Basically, after creating the Kd-tree, 
For every point ![](https://latex.codecogs.com/svg.latex?%7B%5Ccolor%7BBlue%7D%20%5Cboldsymbol%7Bp%7D_i%20%5Cin%20P%7D), the following steps are performed:

1.  Add ![](https://latex.codecogs.com/svg.latex?%7B%5Ccolor%7BBlue%7D%20%5Cboldsymbol%7Bp%7D_i%7D) to the current queue Q;

2. For every point ![](https://latex.codecogs.com/svg.latex?%7B%5Ccolor%7BBlue%7D%20%5Cboldsymbol%7Bp%7D_i%20%5Cin%7D) Q:
    - Search for the set ![](https://latex.codecogs.com/svg.latex?%7B%5Ccolor%7BBlue%7D%20P%5Ei_k%7D) of point neighbors of ![](https://latex.codecogs.com/svg.latex?%7B%5Ccolor%7BBlue%7D%20%5Cboldsymbol%7Bp%7D_i%7D) in a sphere with radius [image];

     - For every neighbor ![](https://latex.codecogs.com/svg.latex?%7B%5Ccolor%7BBlue%7D%20%5Cboldsymbol%7Bp%7D%5Ek_i%20%5Cin%20P%5Ek_i%7D), check if the point has already been processed, and if not add it to Q;
    - When the list of all points in Q has been processed, add Q to the list of clusters C, and reset Q to an empty list

the algorithm terminates when all points ![](https://latex.codecogs.com/svg.latex?%7B%5Ccolor%7BBlue%7D%20%5Cboldsymbol%7Bp%7D_i%20%5Cin%20P%7D) have been processed and are now part of the list of point clusters. 

Hope that helps. I will update the README to make it clear.

praveen-palanisamy, 2020-04-09:
Closing this since the question was addressed. Please feel free to re-open if you have a follow-up or open a new issue.

closed 2020-04-09 by praveen-palanisamy