praveen-palanisamy/multiple-object-tracking-lidar #22: Unstable cluster positions

jonathan84clark, 2020-04-13:
Hello,

I am looking at your really cool object tracking library and I am having some trouble getting it to work with the Lidar I am using. 

My Lidar is the Livox 40 https://www.livoxtech.com/mid-40-and-mid-100

I am using the ROS driver for this driver and re-publishing the PointCloud2 output of that library to the filtered_cloud  topic

What I am seeing is the cubes that represent objects are bouncing around all over the place. I can't see how the position of those cubes relates to things in the real world.

I was wondering if you have ever seen this behavior before? Could this be a configuration issue? Stability problem with my Lidar?

Any help on this would be great!

Thanks,

Jonathan L Clark

praveen-palanisamy, 2020-04-13:
HI @jonathan84clark , I haven't tested with the Livox 40 LIDAR but, my initial hunch (without looking at a sample of the data published to `filtered_cloud` topic) is that the points in the raw point cloud from Livox 40 are spread out in space without any "objectness" and the default point cloud processing and clustering routine is not doing a great job for the scene your LIDAR is looking at.

As a quick workaround, you can filter/pre-process the raw point cloud from Livox 40 before publishing it on to the `filtered_cloud` topic. 

If you can share sample point clouds that you are publishing to `filtered_cloud`, sample outputs etc, it will help to have a better understanding of your problem.

jonathan84clark, 2020-04-13:
Thanks for the help. I will get some Lidar data as soon as possible.

My co-worker suggested I capture several frames of data before sending them to be processed. I want to give that a shot too.

What Lidar did you use? (if you don't mind me asking)

praveen-palanisamy, 2020-04-13:
Sounds good.

This library has been tested with Hokuyo & Velodyne LIDARs (VLP-16, VLP-32) and a bunch of ROS bags with point cloud data that could have been collected from other LIDAR sensors.

jonathan84clark, 2020-04-14:
I attempted to combine the Lidar data, but when I send the data I get this error from the multi-object tracker.
`[pcl::KdTreeFLANN::setInputCloud] Cannot create a KDTree with an empty input cloud!`

I am combining the data with Python and oddly enough Python treats uint8 arrays as strings. I though that simply concatenating the strings would fix the issue but I am wondering if that is where my error is.
`    def lidarCallback(self, data): 
        ms_ticks = rospy.Time.now().to_nsec() * 0.000001
        self.point_cloud_cache.height += 1
        self.point_cloud_cache.data += data.data
        self.point_cloud_cache.fields.extend(data.fields)
        if self.cloud_cache_send < ms_ticks:
            self.point_cloud_cache.point_step = data.point_step
            self.point_cloud_cache.row_step = data.row_step
            self.point_cloud_cache.is_dense = data.is_dense
            self.point_cloud_cache.header = data.header
            self.cloud_proc_publisher.publish(self.point_cloud_cache)
            #print("row_step: " + str(self.point_cloud_cache.row_step) + " Height: " + str(self.point_cloud_cache.height) + " Length: " + str(len(self.point_cloud_cache.data)))
            self.point_cloud_cache.data = ""
            self.point_cloud_cache.fields = []
            self.point_cloud_cache.height = 0
            self.cloud_cache_send = ms_ticks + 500.0`

praveen-palanisamy, 2020-04-14:
> `[pcl::KdTreeFLANN::setInputCloud] Cannot create a KDTree with an empty input cloud!`

From the error message, it certainly is the case that the point cloud that were being published to the `filtered_cloud` is empty. 
Please make sure valid point cloud data is being published on to that topic. You can use `rostopic echo` to quickly verify.

praveen-palanisamy, 2020-06-07:
Closing this as there's no updates. Please re-open or create a new issue if needed.

closed 2020-06-07 by praveen-palanisamy