praveen-palanisamy/multiple-object-tracking-lidar #23: Question on the number of clusters

jhyim5589, 2020-05-14:
First of all, thank you for your great work for object tracking with LiDAR.

I am studying on LiDAR data association with your work and it helps me a lot.

I got a question about the number of clusters that your code publish.

Until now, your code publish only 6 clusters.
In case that # of objects is over 6, modification must be needed.

In my case, I have my own code that process the data to clustering stage.

What is your suggestion about modification that could make the code publish the exact number of clusters matching to the number of clusters.(I came up with for loop that generate the KalmanFilter class, and not sure is it the best way.)

Thank you again for your work and advise in advance

praveen-palanisamy, 2020-05-14:
Hi @jhyim5589 , I'm glad that this repository is useful for your study and work.

This line in the code limits the number of processed clusters to be the top 6 objects:
https://github.com/praveen-palanisamy/multiple-object-tracking-lidar/blob/8c828c1095b58e9af6d92ca17890ebd0bbf29331/src/main.cpp#L281

Instead of `for (int i=0;i<6;i++)` you can do something like `for(int i=0, i< objID.size(); i++)` and go from there to update the code to support more than 6 objects.
Note that, the computation time will increase as the number of objects increases. 

Hope that helps. Feel free to submit a pull request with your modifications for review/integration.

praveen-palanisamy, 2020-06-07:
@jhyim5589 , Closing this since the question was answered and there's no updates. Please re-open or create a new issue if needed.
Please consider submitting a pull request if you would like to integrate your work on extending to more than 6 objects.

closed 2020-06-07 by praveen-palanisamy

jhyim5589, 2020-06-19:
Hi @praveen-palanisamy 

Thank you for your kind reply on my first question.

I am still working on modifying # of kalman filters based on # of clusters.
I have met another problem here, and now ask for your advise.

1. Initiating kalman filter should be outside of the callback.
2. I need # of clusters to initiate exact # of kalman filters.
how can I initiate kalman filters in this case?

your advise will be great help.

thank you so much.

praveen-palanisamy, 2020-06-19:
Hi @jhyim5589 ,
Good to see your continued interest.
1. That's right. The Kalman filter needs to be initialized outside the callback. 
2. This is a standard question in multiple-object detection scenarios. It is rare to know beforehand how many objects will be in the LIDAR scans or in the scene. One way is to limit the maximum number of objects that can be detected and tracked (e.g. 6 as in the current code). The other way is to dynamically create a filter instance and start tracking when new object clusters are detected. For this, you could use a vector of Kalman filters (`std::vector<KalmanFilter> KFs`) and add (`KFs.push_back(...)`) new filters when a new, unassociated cluster is detected. 

Hope that helps.

jhyim5589, 2020-06-19:
Thanks very much.

Actually I am working through python, I am not sure KFs class is offered in python opencv.
Still your advise is greatly helped.

Thank you so much.

ingramator, 2021-06-23:
It would be create if the number of clusters could expand based on new input clusters, with new KFs being initialised upon new cluster detection. 

If anyone has made a fork with this please let me know or I will try implement it myself (in cpp)!