praveen-palanisamy/multiple-object-tracking-lidar #27: How to publish las file to filtered cloud?

SkylerZheng45, 2020-07-02:
Thanks for the great project!
Since the project supports point cloud datasets, may I ask how I can publish my las file to filtered cloud? I'm just totally lost about how to make the project run on the las file.

praveen-palanisamy, 2020-07-02:
Hi @SkylerZheng45  !

You could run a ROS node that reads the pointcloud data from your LAS file and publishes it to the `filtered_cloud` topic.

Another way that I can think of is to convert the LAS file into [PCD file](http://pointclouds.org/documentation/tutorials/pcd_file_format.html) and use the [pcd_to_pointcloud](http://wiki.ros.org/pcl_ros#pcd_to_pointcloud) node to publish the data.

Hope that helps.

praveen-palanisamy, 2020-11-18:
Closing this as a couple of resolution methods were proposed. Please re-open if more help is needed.

closed 2020-11-18 by praveen-palanisamy