praveen-palanisamy/multiple-object-tracking-lidar #28: Closing the issue as the question was answered. Reopen if you need more info. Thanks!

chumingqian, 2020-07-29:
Closing the issue as the question was answered. Reopen if you need more info. Thanks!

_Originally posted by @praveen-palanisamy in https://github.com/praveen-palanisamy/multiple-object-tracking-lidar/issues/9#issuecomment-525173447_

chumingqian, 2020-07-29:
Hi,palanisamy:
     @praveen-palanisamy  thanks  for  your sharing,  and  i  have a question.  
  In  src/main.cpp    line 21 :  #include <sensor_msgs/PointCloud2.h>
 data type  is  pointCloud2.msg.
 However , my  data type  is  a  LaserScan.msg.   When  i  run the code, i got the  error sth  about  datatype  mistake,   how should  i  solve  this , could  you  give  a  guide , thanks in advance .

[image]
[image]

praveen-palanisamy, 2020-07-30:
Hi @chumingqian , 
You can use the [laser_geometry](http://wiki.ros.org/laser_geometry) ROS package to convert your `LaserScan` messages to `PointCloud2` message.
Here's a code example:

```c++
laser_geometry::LaserProjection projector_;

void scanCallback (const sensor_msgs::LaserScan::ConstPtr& scan_in)
{
  sensor_msgs::PointCloud2 point_cloud;
  projector_.projectLaser(*scan_in, point_cloud);

  // You can now use point_cloud
}
```

closed 2020-07-30 by chumingqian

chumingqian, 2020-07-30:
> Hi @chumingqian ,
> You can use the [laser_geometry](http://wiki.ros.org/laser_geometry) ROS package to convert your `LaserScan` messages to `PointCloud2` message.
> Here's a code example:
>
> ```c++ [...]

It  worked  ,thank  you  so  much!!!