praveen-palanisamy/multiple-object-tracking-lidar #29: cant launch node, assertion error

Ahmad-Drak, 2020-08-10:
Hi, I seem to be getting this error when launching the node. Any ideas?

> About to setup callback
> kf_tracker: /build/pcl-6_P28C/pcl-1.7.2/kdtree/include/pcl/kdtree/impl/kdtree_flann.hpp:172: int pcl::KdTreeFLANN<PointT, Dist>::radiusSearch(const PointT&, double, std::vector<int>&, std::vector<float>&, unsigned int) const [with PointT = pcl::PointXYZ; Dist = flann::L2_Simple<float>]: Assertion `point_representation_->isValid (point) && "Invalid (NaN, Inf) point coordinates given to radiusSearch!"' failed.
> Aborted (core dumped)

praveen-palanisamy, 2020-08-11:
Hi @Ahmad-Drak , what is the source of the point clouds you are using? The point cloud seems to contain NaN/Inf values.

Please make sure that the input point clouds doesn't have any invalid data and then try again and report back the GDB trace if you still run into an issue.

Ahmad-Drak, 2020-08-11:
its a bag file that publishes point cloud to a topic and i remap the topic to `/filtered_cloud`. I checked `rostopic echo /filtered_cloud` and i dont see any Nan/Inf values (of course the data is huge and i could have easily missed it).
I read around and others seem to have a reason to believe that this error is related to pcl_ros package?

praveen-palanisamy, 2020-08-11:
Hmmm. The error is definitely coming from the PCL library (not from the code in this repository).
I still suspect that there is some invalid data in the point clouds (from the bag file). Remapping and publishing to the `/filtered_cloud` topic should be fine as long as the point cloud is really filtered to remove invalid points. Even one or a few invalid point cloud data can cause this issue. 
It's better to filter the point clouds after loading and before publishing on to the `filtered_cloud` topic.

You can write your routine to remove the `NaN`  (or other invalid data) or try using the PCL library with something like this:

```c++
std::shared_ptr<std::vector<int>> indices(new std::vector<int>);
pcl::removeNaNFromPointCloud(*source_cloud, *indices);
pcl::ExtractIndices<pcl::PointXYZ> extract;
extract.setInputCloud(source_cloud);
extract.setIndices(indices);
extract.setNegative(true);
extract.filter(*source_cloud);
```

praveen-palanisamy, 2020-08-20:
@Ahmad-Drak , is the issue resolved on your end?

praveen-palanisamy, 2020-09-11:
Closing this since the solution proposed in https://github.com/praveen-palanisamy/multiple-object-tracking-lidar/issues/29#issuecomment-672098760 seems to fix the issue with NaN values causing the error as reported by others (see referenced issues above).

closed 2020-09-11 by praveen-palanisamy

Tonyc48, 2021-04-04:
Hello, I am getting the same error after running the node:
"About to setup callback"

Although, it does not give me an error for invalid NaN points.  It just continuously runs until I kill the command.  Also, I am running this on Raspbian rather than Ubuntu.  I am not sure if it is the PCL libraries but I am having some trouble.  Any help would be appreciated.

Ahmad-Drak, 2021-04-06:
Sorry @Tonyc48 I haven't worked on this much, had to change directions. Im sure @praveen-palanisamy can be better help

amineabm, 2022-01-07:
> Hello, I am getting the same error after running the node: "About to setup callback"
>
> Although, it does not give me an error for invalid NaN points. It just continuously runs until I kill the [...]

Hello @Tonyc48, Did you found a solution for this issue ?

harshmay, 2023-04-07:
Hi there, I was confused about where and how you were using the filtered NaN cloud in the code as I have two ways to get the data. These are either using the ROSBAG to save point cloud in a .bag file or get live data points for the LiDAR in real time so that we can run the tracker in real time.