praveen-palanisamy/multiple-object-tracking-lidar #33: how to  visualize viz using RViz

RussellGee, 2020-09-08:
my data source is rosbag from KITTI velodyne_points. step by the usage, i have got the follow out put:

$ rosrun multi_object_tracking_lidar kf_tracker

About to setup callback
filterN=0
filterN=1
filterN=2
filterN=3
filterN=4
filterN=5
distMat.size()6
distMat[0].size()6
12.5153 12.1626 20.0648 10.9252 6.92213 26.0863 
12.5153 12.1626 20.0648 10.9252 6.92213 26.0863 
12.5153 12.1626 20.0648 10.9252 6.92213 26.0863 
12.5153 12.1626 20.0648 10.9252 6.92213 26.0863 
12.5153 12.1626 20.0648 10.9252 6.92213 26.0863 
12.5153 12.1626 20.0648 10.9252 6.92213 26.0863 
findIndexOfMin cALLED
minEl=3.40282e+38
minIndex=0,4
Received minIndex=0,4
clusterCount=0
findIndexOfMin cALLED
minEl=3.40282e+38
minIndex=1,3
Received minIndex=1,3
clusterCount=1
findIndexOfMin cALLED
minEl=3.40282e+38
minIndex=2,1
Received minIndex=2,1
clusterCount=2
findIndexOfMin cALLED
minEl=3.40282e+38
minIndex=3,0
Received minIndex=3,0
clusterCount=3
findIndexOfMin cALLED
minEl=3.40282e+38
minIndex=4,2
Received minIndex=4,2
clusterCount=4
findIndexOfMin cALLED
minEl=3.40282e+38
minIndex=5,5
Received minIndex=5,5
clusterCount=5
filterN=0
filterN=1
filterN=2
filterN=3
filterN=4
filterN=5
distMat.size()6
distMat[0].size()6
11.5807 5.98043 6.29488 10.3126 25.6532 12.725 
11.3332 5.60705 5.95035 9.93752 25.2852 12.6304 
11.0587 5.68 5.90826 10.0085 25.6847 12.1575 
11.4436 6.4903 6.62139 10.8004 26.6076 12.1298 
13.9819 8.2814 8.69094 12.5877 26.9738 14.7729 
11.477 4.91065 5.55812 9.11519 23.712 13.4483 
findIndexOfMin cALLED
minEl=3.40282e+38
minIndex=5,1
Received minIndex=5,1
clusterCount=0
findIndexOfMin cALLED
minEl=3.40282e+38
minIndex=2,2
Received minIndex=2,2
clusterCount=1
findIndexOfMin cALLED
minEl=3.40282e+38
minIndex=1,3
Received minIndex=1,3
clusterCount=2
findIndexOfMin cALLED
minEl=3.40282e+38
minIndex=3,0
Received minIndex=3,0
clusterCount=3
findIndexOfMin cALLED
minEl=3.40282e+38
minIndex=0,5
Received minIndex=0,5
clusterCount=4
findIndexOfMin cALLED
minEl=3.40282e+38
minIndex=4,4
Received minIndex=4,4
clusterCount=5
filterN=0
filterN=1
filterN=2
filterN=3
filterN=4
filterN=5
distMat.size()6
...
------------------------------------------------------------------------------------------------------------------
and the cammand " rostopic echo /viz" :
 $ rostopic echo /viz 
markers: 
  - 
    header: 
      seq: 0
      stamp: 
        secs: 0
        nsecs:         0
      frame_id: "/map"
    ns: ''
    id: 0
    type: 1
    action: 0
    pose: 
      position: 
        x: 4.52638244629
        y: 5.33555030823
        z: 6.87136125634e-05
      orientation: 
        x: 0.0
        y: 0.0
        z: 0.0
        w: 0.0
    scale: 
      x: 0.3
      y: 0.3
      z: 0.3
    color: 
      r: 0.0
      g: 0.0
      b: 0.0
      a: 1.0
    lifetime: 
      secs: 0
      nsecs:         0
    frame_locked: False
    points: []
    colors: []
    text: ''
    mesh_resource: ''
    mesh_use_embedded_materials: False
  - 
    header: 
      seq: 0
      stamp: 
        secs: 0
        nsecs:         0
      frame_id: "/map"
    ns: ''
    id: 1
    type: 1
    action: 0
    pose: 
      position: 
        x: -3.29236984253
        y: 16.9774074554
        z: -8.0042569607e-05
      orientation: 
        x: 0.0
        y: 0.0
        z: 0.0
        w: 0.0
    scale: 
      x: 0.3
      y: 0.3
      z: 0.3
    color: 
      r: 1.0
      g: 1.0
      b: 1.0
      a: 1.0
    lifetime: 
      secs: 0
      nsecs:         0
    frame_locked: False
    points: []
    colors: []
    text: ''
    mesh_resource: ''
    mesh_use_embedded_materials: False
  - 
    header: 
      seq: 0
      stamp: 
        secs: 0
        nsecs:         0
      frame_id: "/map"
    ns: ''
    id: 2
    type: 1
    action: 0
    pose: 
      position: 
        x: 8.20040512085
        y: 8.98066139221
        z: 2.25422081002e-08
      orientation: 
        x: 0.0
        y: 0.0
        z: 0.0
        w: 0.0
    scale: 
      x: 0.3
      y: 0.3
      z: 0.3
    color: 
      r: 0.0
      g: 1.0
      b: 1.0
      a: 1.0
    lifetime: 
      secs: 0
      nsecs:         0
    frame_locked: False
    points: []
    colors: []
    text: ''
    mesh_resource: ''
    mesh_use_embedded_materials: False
  - 
    header: 
      seq: 0
      stamp: 
        secs: 0
        nsecs:         0
      frame_id: "/map"
    ns: ''
    id: 3
    type: 1
    action: 0
    pose: 
      position: 
        x: -1.16644775867
        y: -9.15706729889
        z: 0.000111785302579
      orientation: 
        x: 0.0
        y: 0.0
        z: 0.0
        w: 0.0
    scale: 
      x: 0.3
      y: 0.3
      z: 0.3
    color: 
      r: 1.0
      g: 0.0
      b: 1.0
      a: 1.0
    lifetime: 
      secs: 0
      nsecs:         0
    frame_locked: False
    points: []
    colors: []
    text: ''
    mesh_resource: ''
    mesh_use_embedded_materials: False
  - 
    header: 
      seq: 0
      stamp: 
        secs: 0
        nsecs:         0
      frame_id: "/map"
    ns: ''
    id: 4
    type: 1
    action: 0
    pose: 
      position: 
        x: 0.540309131145
        y: 6.70243358612
        z: 7.79212678026e-06
      orientation: 
        x: 0.0
        y: 0.0
        z: 0.0
        w: 0.0
    scale: 
      x: 0.3
      y: 0.3
      z: 0.3
    color: 
      r: 0.0
      g: 1.0
      b: 0.0
      a: 1.0
    lifetime: 
      secs: 0
      nsecs:         0
    frame_locked: False
    points: []
    colors: []
    text: ''
    mesh_resource: ''
    mesh_use_embedded_materials: False
  - 
    header: 
      seq: 0
      stamp: 
        secs: 0
        nsecs:         0
      frame_id: "/map"
    ns: ''
    id: 5
    type: 1
    action: 0
    pose: 
      position: 
        x: 3.1932387352
        y: 6.08556127548
        z: -7.27250744603e-06
      orientation: 
        x: 0.0
        y: 0.0
        z: 0.0
        w: 0.0
    scale: 
      x: 0.3
      y: 0.3
      z: 0.3
    color: 
      r: 1.0
      g: 1.0
      b: 1.0
      a: 1.0
    lifetime: 
      secs: 0
      nsecs:         0
    frame_locked: False
    points: []
    colors: []
    text: ''
    mesh_resource: ''
    mesh_use_embedded_materials: False
----------------------------------------------------------------------------------------
i am not sure whether Is the result correct or not

RussellGee, 2020-09-08:
i have knew how to visualize viz.

closed 2020-09-08 by RussellGee

Surajj4837-th, 2021-03-08:
I m also trying to visualise the output, will you @praveen-palanisamy  or @RussellGee elaborate the process. The topics are appearing in RViz but how to see the final output?

[image]