praveen-palanisamy/multiple-object-tracking-lidar #35: VLP 32C Data Interpertation

SHamblin, 2020-09-10:
Thanks for the work you've put into the project! I've got a question about using it with a VLP 32C. I'm attempting to detect a car in a parking log and the clusters that are being picked up don't seem to match what I would expect.

I'm playing back the data from a bag file and remapping to the filtered_cloud topic. In the shots of Rviz bellow you can see the input data and then in white the detected clusters. 

Is there a variable that I should adjust? 
`<ec.setClusterTolerance (0.08); >` Seems like a promising line to adjust but I'm unsure if I should increase or lower it.

[image]
[image]

Any help would be appreciated, Thank you!

praveen-palanisamy, 2020-09-11:
Hi @SHamblin , 
In your use-case, is the car that you are intending to detect & track stationary or moving (w.r.t the VLP 32C)?
For this tracker to work without modifications, the point cloud input is expected to be "*filtered*" before publishing onto the `filtered_cloud` topic.
You could pre-process your LiDAR point clouds to remove the stationary points belonging to the background (concentric lines etc) and then publish to the `filtered_cloud` topic to feed this node.

praveen-palanisamy, 2020-11-18:
Closing this as the question was answered. Please re-open if more information/clarification is needed.

closed 2020-11-18 by praveen-palanisamy