praveen-palanisamy/multiple-object-tracking-lidar #36: is the /filtered_cloud 2D or 3D?

abhaysjuneja, 2020-09-22:
I tried to run the node with VLP-16 point cloud (not filtered). Are we supposed to provide it a 2D cloud (top-down)? or do you expect it to work fine with a 3D point cloud? 
And when you say filtered, do you mean a 3D cloud with ground point filtered out?
[image]

praveen-palanisamy, 2020-09-25:
Hi @abhaysjuneja ,
Good to hear that you have been able to use this package with data from VLP-16. The algorithm relies on planar segmentation so, it operates on a plane. The input point cloud can be 3D as long as the objects of interest for you are on the same plane as the robot/car/sensor and there exists at least one surface on the objects of interest that all intersect with the plane of segmentation.
It will work for point clouds from LiDARs/lasers mounted on ground robots like a Car or a Truck for example.

This package assumes that the point cloud has been pre-processed to remove background information like trees, facade etc which are usually not relevant for tracking. Such a pre-processed point cloud is what is referred to as "filtered"

Hope that helps.

praveen-palanisamy, 2020-11-18:
Closing this as the question was answered. Please re-open if more information is needed.

closed 2020-11-18 by praveen-palanisamy