praveen-palanisamy/multiple-object-tracking-lidar #41: can not track small dynamic obstacles when using a single-line laser

UyaSong, 2021-01-29:
Hi @praveen-palanisamy , thanks for sharing your work, it's quite helpful to me.
I want to run this pkg on a mobile robot with a single-line laser, but it can hardly detect the dynamic obstacle (human legs), like the video shows.
[image]
I don't know whether I had used it correctly. Maybe I should increase the number of cluster and kalman filter. Or should I use the featureDetection class?

praveen-palanisamy, 2021-01-30:
Hi @UyaSong ,
Good to hear that it is helpful for you.
From the GIF you shared, it looks like there are objects other than the human-legs (that you are interested in) that are bigger in size (number of points in the point cloud) and appear to be moving. This could be due to noise in your laser sensor input that haven't been filtered before publishing on to the `filtered_cloud` topic. You could try filtering those noises out since they appear to be part of the scene and may not be really moving objects.

You can also increase the number of clusters so that you can have more objects detected and tracked which will likely include the (smaller) human-leg point clouds.

UyaSong, 2021-01-31:
Hi @praveen-palanisamy , 
Thanks for your reply.
The effect of GIF in readme is quite ideal for me. What filter did you use there to remove the surrounding environment point cloud?

praveen-palanisamy, 2021-02-01:
It's been quite some time since then that I am not able to recall all the details but the implementation was using [PCL](https://pointclouds.org/). Specifically the [pcl_filters](https://pointclouds.org/documentation/group__filters.html) ([pcl_segmentation](https://pointclouds.org/documentation/group__segmentation.html) also is useful for filtering background planes).

There's a tutorial demonstrating the use of the [`pcl_filters` for outlier removal here](https://pcl.readthedocs.io/projects/tutorials/en/latest/remove_outliers.html) which you may find useful. The `pcl_ros` package has ROS nodes implementing those outlier removal methods. More info here: [http://wiki.ros.org/pcl_ros/Tutorials/filters](http://wiki.ros.org/pcl_ros/Tutorials/filters). 

I hope that helps you filter the (static/baground/noisy) point clouds.

UyaSong, 2021-02-01:
Thanks a lot! I will have a try.

mshmsh1512987, 2022-05-23:
> Hi @praveen-palanisamy , Thanks for your reply. The effect of GIF in readme is quite ideal for me. What filter did you use there to remove the surrounding environment point cloud?

hello,I have some questions in Point cloud topic release and acceptance，Can you add me? Pay for some advice，qq:2335702163

FortuneYU, 2022-05-29:
> 

> hello,I have some questions in Point cloud topic release and acceptance，Can you add me? Pay for some advice，qq:2335702163

Hi @mshmsh1512987   , you can refer to below link for point cloud topic subscribe and publish. It works well.  @mshmsh1512987 
https://industrial-training-master.readthedocs.io/en/melodic/_source/session5/Building-a-Perception-Pipeline.html

jk-ethz, 2022-06-22:
> Hi @praveen-palanisamy , Thanks for your reply. The effect of GIF in readme is quite ideal for me. What filter did you use there to remove the surrounding environment point cloud?

@UyaSong Hi, I have a similar use case. Did you figure out how to efficiently filter the scans for use with this package?

jk-ethz, 2022-07-12:
I have developed a filter:
https://github.com/jk-ethz/laser_static_map_filter