praveen-palanisamy/multiple-object-tracking-lidar #43: What are the classes in which the code classifies the objects? Any accuracy metric available? What is the speed you are getting?

Surajj4837-th, 2021-03-10:
What are the classes in which the code classifies the objects? Any accuracy metric available? What is the speed you are getting?

praveen-palanisamy, 2021-04-23:
Hi @Surajj4837-th. Missed this question from you until now. Please see my answers below:

> What are the classes in which the code classifies the objects? Any

Any static/dynamic object is detected and tracked. The clustering of pointclouds into an object happens in an unsupervised way so there's no "class labels" associated with them.

> Any accuracy metric available?

No. It's been mainly used on live pointcloud streams and wasn't benchmarked on a static dataset which has annotations to get such metrics.

>What is the speed you are getting?

IIRC, the results were published >=30FPS. It depends on the number of points in the pointcloud and the hardware this is running

Surajj4837-th, 2021-04-24:
Thanks for the update.

closed 2021-04-24 by Surajj4837-th

boazMgm, 2022-01-27:
Hi,
Is there a way to track only dynamic objects?
Thanks a lot.