praveen-palanisamy/multiple-object-tracking-lidar #47: ROS node hangs when running

FixedSpecialty, 2021-11-08:
I saw a similar issue and followed all the steps/read the articles [Understanding ROS Topics](http://wiki.ros.org/ROS/Tutorials/UnderstandingTopics) and [Introduction to Working With Laser Scanner Data](http://wiki.ros.org/laser_pipeline/Tutorials/IntroductionToWorkingWithLaserScannerData). The issue I am having is that even after publishing the bag file to the filtered_cloud node,`rosrun multi_object_tracking_lidar kf_tracker` still hangs. The bag file's topic publishes to /livox/lidar so I am "forwarding" it to /filtered_cloud.

I am running in separate terminals on Windows 10 in Ubuntu 18.04 LTS terminals:
 `roscore`
`rosrun multi_object_tracking_lidar kf_tracker` - After running this it hangs on `About to setup callback` forever!
`rosbag play --clock /mnt/d/RosBags/demo2.bag /livox/lidar:=/filtered_cloud`
`rosparam set use_sim_time true` and `rosrun tf static_transform_publisher 0 0 0 0 0 0 map dummy_frame 100`

After all these commands, and publishing I am stuck on the `About to setup callback` message. Any help is greatly appreciated!

The image below shows my RVIZ setup as well.
[image]

FixedSpecialty, 2021-11-09:
Delay was caused by a lack of processing power. When pausing the rosbag halfway through while everything is running. The kf_tracker would actually run around 2 minutes later.

closed 2021-11-09 by FixedSpecialty

praveen-palanisamy, 2021-11-11:
Hi @FixedSpecialty , Thanks for summarizing the issue you faced and following it up with the reason and resolution. It's going to be helpful for the community. I appreciate it!