praveen-palanisamy/multiple-object-tracking-lidar #48: I can't see /filtered_cloud on RVIZ

Constellation6, 2021-11-29:
Thanks for sharing the source. 
I want to use /filtered_cloud but i can't see /filtered_cloud on RVIZ and I don't know how publish topic.
Please give me a solution
[image]
.

tilexiadam, 2022-02-18:
Any progress on this? If so, can you please share the solution?

aldras, 2022-05-19:
filtered_cloud is the topic that this package subscribes to, it does not create it (I don't think).
Please either publish your cloud data on that topic (if you can) or change the topic that this package subscribes to for the filtered_cloud.  This is located in main.cpp, line 687 (as of commit [a9aa364](https://github.com/praveen-palanisamy/multiple-object-tracking-lidar/commit/a9aa364d334933a099abf998d3f4f061f1949d00) on Aug 29, 2021.  The below line, change "filtered_cloud" to a topic of your choice and recompile.
`  // Create a ROS subscriber for the input point cloud
  ros::Subscriber sub = nh.subscribe("filtered_cloud", 1, cloud_cb);`

praveen-palanisamy, 2022-05-23:
Hi @Constellation6 , @tilexiadam , @aldras : Sorry for the delay. I hope  you have already resolved the issue on your end if not, please see this wiki page for example on how to publish to `/filtered_cloud` ROS topic: [Sensor setup and publishing to fileted_cloud](https://github.com/praveen-palanisamy/multiple-object-tracking-lidar/wiki/velodyne_vlp16).

Specifically, for your setup, you can remap the topics so that the data gets published on to the `/filtered_cloud` topic by using this line in your `roslaunch` file:

`<remap from="scan_2D" to="filtered_cloud"/>`

Visit [this](https://wiki.ros.org/roslaunch/XML/remap) page for information on the remap tag. You can also use the command-line topic mapping like this:

`rosrun sensor_pkg_name sensor_node_name scan_2D:=filtered_cloud`

Hope that helps.

praveen-palanisamy, 2022-05-23:
Closing this since the question has been answered. Feel free to post if you still need help.

closed 2022-05-23 by praveen-palanisamy