praveen-palanisamy/multiple-object-tracking-lidar #51: How to publish the RPLidar Data to the Filtered_Cloud rostopic

tilexiadam, 2022-02-18:
Thank you for this source and sorry if this question has already been answered but I could not manage to do this. 

I am using an Slamtec RPlidar and here is a screenshot from RViz:

[image]

The problem is, I couldn't figure how to publish the rplidar scan data onto the filtered_data.

aldras, 2022-05-19:
Alter the topic, in the above mentioned "issue" and change it to "scan" as I see in your image.
This should grab the point cloud data from your device, if it is indeed outputting on the "scan" topic.

praveen-palanisamy, 2022-05-23:
@aldras : Thank you for your comments!

@tilexiadam : If you are still facing this issue, please try [this solution](https://github.com/praveen-palanisamy/multiple-object-tracking-lidar/issues/48#issuecomment-1134177467). You can rename your existing pointcloud topic (`scan` from your screenshot) to `filtered_cloud` using one of the following:
1. In roslaunch file: `<remap from="scan" to="filtered_cloud"/>`
1. On the command-line: `rosrun sensor_pkg_name sensor_node_name scan:=filtered_cloud`

praveen-palanisamy, 2022-05-23:
Closing this since the question has been answered. Feel free to post if you still need help.

closed 2022-05-23 by praveen-palanisamy