praveen-palanisamy/multiple-object-tracking-lidar #52: ROS 2 Foxy Fitzroy port

aldras, 2022-05-19:
I would like to share the port that I did of this to ROS 2 Foxy Fitzroy.
I have spent the past several few days understanding how this works, and have a good enough understanding to port it over.

I am currently attempting to use a Livox LiDAR Mid-70 with this for object detection and tracking.

I am not 100% everything works properly, and there are a few things that need work.
For instance there is no MarkerArray support in ROS2 that I am aware of, I spent hours trying to get that working.
Also Marker is not working correctly with a topic subscription so it is commented out.

In src/multiple_object_tracking_lidar.cpp I had to change lines 405 and 547 from "ec.setClusterTolerance(0.3)" to "ec.setClusterTolerance(0.08)" to resolve some speed issues.

As far as it functioning correctly, I am not 100% sure yet as I just finished the port and it compiles.
I am planning to use this as a base for a project to track objects.

I am not sure how much I can share after I put a lot of time into it since it will become more of a company project, however, anything that I can share and update I would be willing.

Initially I wanted to preserve as much as I could, with no bug fixes or additions, just a port to ROS 2 Foxy Fitzroy.

As a lot has changed I am not sure I can do a pull request, I basically created a new ROS 2 project from scratch and copy/pasted the code in as I ported it.  The functionality should be identical with the addition of a namespace and a class.

In the launch file the filtered_cloud as well as the frame_id can be set, I would have added more to the launch file, such as auto launching rviz2, but wanted to keep this as basic as possible.

Please feel free to fork/fix and use the below code as you please, or create a new branch or repository to hold it.
There is no license in the header of any of the files, I have never been too good about knowing what to copy/paste there, but, it can remain the same license as the original.

https://github.com/aldras/multiple_object_tracking_lidar_ros2.git

aldras, 2022-05-19:
The code was ported using commit 27db548 Aug 30, 2021.

praveen-palanisamy, 2022-05-21:
Hi @aldras ,
Great to see your summary and port of this project to ROS 2 Foxy Fitzroy. You mentioned a lot of points in our post!
It is harder for me to review the code changes since you have it on a separate, disconnected repository. If you decide to make a pull request (or apply your patch to a fork), it will be appreciated.