praveen-palanisamy/multiple-object-tracking-lidar #54: Can this pkg be used in Ros kinetic 16.04?

Delbina, 2022-08-04:
Hi,

I am working with Ros kinetic 16.04. does this pkg work on Ros kinetic?

thanks

praveen-palanisamy, 2022-08-05:
Hi @Delbina ,
Yes, this package is expected to work with ROS Kinetic. Here's the [link to Multi Object Tracking LiDAR ROS pkg index page for Kinetic](https://index.ros.org/p/multi_object_tracking_lidar/github-praveen-palanisamy-multiple-object-tracking-lidar/#kinetic).

Please [open a discussion](https://github.com/praveen-palanisamy/multiple-object-tracking-lidar/discussions) if you have any follow-up question. Thank you!

closed 2022-08-05 by praveen-palanisamy

Delbina, 2022-08-06:
thanks a lot, 
just my point clouds published on to the /velodyne_points topic. where can i change filtered_cloud to /velodyne_points ??

Delbina, 2022-08-06:
> Hi @Delbina , Yes, this package is expected to work with ROS Kinetic. Here's the [link to Multi Object Tracking LiDAR ROS pkg index page for [...]

thanks a lot, 
i did remapping from /velodyne_points topic to filtered_cloud . but it is not publishing! i donot know why?! would you please help me ?>