praveen-palanisamy/multiple-object-tracking-lidar #59: Results not aligned with the rest of the map and with scans

Marabir, 2023-05-25:
I run turtlebot simulation with amcl localization in gazebo, change laser scan to pointcloud with code attached below. 

filtered_cloud which I got is well aligned in rviz, but the result of the kf_tracker is not. It is shifted as it does not get proper localization from amcl or tf tree and when I ordered robot to move around, it rotates in the opposite direction as the robot does.

[image]

where maybe  better visible
[image]

this is the code which I use to transform scans to pointclouds, but as I mentioned, filtered_cloud is aligned well in rviz :/ 

```
#include "ros/ros.h"
#include "sensor_msgs/LaserScan.h"
#include "sensor_msgs/PointCloud2.h"
#include "laser_geometry/laser_geometry.h"
#include "tf/transform_listener.h"

int main(int argc, char** argv){
    
    ros::init(argc, argv, "laser_scan_to_pointcloud");

    ros::NodeHandle nh;
    laser_geometry::LaserProjection projector;
    tf::TransformListener listener_;
    

    ros::Publisher pub = nh.advertise<sensor_msgs::PointCloud2>("/filtered_cloud",1);

    ros::Subscriber sub = nh.subscribe<sensor_msgs::LaserScan>("/scan", 1, [&](const sensor_msgs::LaserScan::ConstPtr& scan) {
        
        if(!listener_.waitForTransform(
            scan->header.frame_id,
            "/base_link",
            scan->header.stamp + ros::Duration().fromSec(scan->ranges.size()*scan->time_increment),
            ros::Duration(1.0))){
                return;
            }
        
        
        // Convert the laser scan to a point cloud
        sensor_msgs::PointCloud2 cloud;
        projector.transformLaserScanToPointCloud("base_link",*scan, cloud,listener_);

        // Publish the point cloud
        pub.publish(cloud);
    });

    ros::spin();

   return 0;
}
```

I will be grateful for your help :smiley:

wangsh386, 2023-05-27:
Hello, could you please share your point cloud file? We would like to try the effect you did, and we would greatly appreciate it  1600475494@qq.com

Marabir, 2023-05-31:
Hi, I've sent you an email with rosbag containing important topics :smiley:

Jakubach, 2025-04-03:
Hi @Marabir @wangsh386,
did you resolve that issue? I faced the same problem